kcjengr/probe_basic #7: Question regarding manual tool change

mark0815, 2020-02-11:
Hi.

I've just added probe_basic to my mill controller and it looks pretty awesome!

While looking into the repo I noticed that the `change.ngc` macro was removed recently. I just wonder how the manual tool change (I have no atc attached to my mill) will work in future as my understanding is that the `change.ngc` in conjunction with the `CHANGE_POSITION` from the `.ini` takes care of the spindle position on manual tool change, am I right?

Regards
Mark

Lcvette, 2020-02-11:
this was the change.ngc file which seemed to be more of a tool length setter macro tied to my old gmoccapy screenset.. i could be wrong, but im not sure this would be needed or more accurately soemthing more universal probably needs to be put in place of this. since it was previously customized for gmoccapy and tool length offset routines..

```
o<change> sub
;(debug, in change tool_in_spindle=#<tool_in_spindle> my_current_pocket=#<my_current_pocket>)
;(debug, selected_tool=#<selected_tool> selected_pocket=#<selected_pocket>)

;otherwise after the M6 this information is gone!
#<tool> = #<selected_tool>
#<pocket> = #<selected_pocket>

; we must execute this only in the milltask interpreter
; or preview will break, so test for '#<_task>' which is 1 for 
; the milltask interpreter and 0 in the UI's
O100 if [#<_task> EQ 0]
        (debug, Task ist Null)
O100     return [999]
O100 endif

;first go up
G53 G0 Z[#<_ini[CHANGE_POSITION]Z>]
; then move to change position
G53 G0 X[#<_ini[CHANGE_POSITION]X>] Y[#<_ini[CHANGE_POSITION]Y>]

; cancel tool offset
G49

; using the code being remapped here means 'use builtin behaviour'
M6

O200 if [#<_hal[gmoccapy.toolmeasurement]> EQ 0]
O200 return [3] ; indicate no tool measurement 
O200 endif

G53 G0 X[#<_ini[TOOLSENSOR]X>] Y[#<_ini[TOOLSENSOR]Y>]
G53 G0 Z[#<_ini[TOOLSENSOR]Z>]

O300 if [#<_hal[gmoccapy.searchvel]> LE 0]
O300 return [-1] ; indicate searchvel <= 0 
O300 endif

O400 if [#<_hal[gmoccapy.probevel]> LE 0]
O400 return [-2] ; indicate probevel <= 0 
O400 endif

F #<_hal[gmoccapy.searchvel]>
G91
G38.2 Z #<_ini[TOOLSENSOR]MAXPROBE>
G0 Z2
; This is commented out only for sim.
;F #<_hal[gmoccapy.probevel]>
;G38.2 Z-4

O500 if [#5070 EQ 0]
G90
O500 return [-3] ; indicate probe contact failure to epilog
O500 endif

G90
G53 G0 Z[#<_ini[CHANGE_POSITION]Z>]

#<touch_result> = #5063
#<probeheight> = #<_hal[gmoccapy.probeheight]> 
#<blockheight> = #<_hal[gmoccapy.blockheight]>

;(DEBUG, #<touch_result>  #<probeheight>  #<blockheight>)

G10 L1 P#<tool> Z[#<touch_result> - #<_hal[gmoccapy.probeheight]> + #<_hal[gmoccapy.blockheight]>]
G43

;G10 L1 P#<tool> Z#<touch_result>
;G10 L2 P0 Z[#<workpieceheight> + #<probeheight> + #<touch_result>]

; signal success be returning a value > 0:
o<change> endsub [1]'''

Lcvette, 2020-02-12:
> Hi.
>
> I've just added probe_basic to my mill controller and it looks pretty awesome!
>
> While looking into the repo I noticed that the `change.ngc` macro was removed recently. I just wonder how the [...]

for manual tool change you would set the ini settings in the [EMCIO] section, as in the docs here:

http://linuxcnc.org/docs/2.8/html/config/ini-config.html#_emcio_section

TOOL_CHANGE_POSITION = 0 0 2 - Specifies the XYZ location to move to when performing a tool change if three digits are used. Specifies the XYZABC location when 6 digits are used. Specifies the XYZABCUVW location when 9 digits are used. Tool Changes can be combined. For example if you combine the quill up with change position you can move the Z first then the X and Y. 

and

TOOL_CHANGE_QUILL_UP = 1 - The Z axis will be moved to machine zero prior to the tool change when the value is 1. This is the same as issuing a G0 G53 Z0.

hope that helps!

closed 2020-02-12 by Lcvette

mark0815, 2020-02-12:
Hi.
Thanks for the response!
I've seen those entries in the `ini` and in the `change.ngc`. That's the reason I thought the manual tool change will no longer work after the removal of the `change.ngc`.
But I guess that's not true and the manual tool change with the TOOL_CHANGE_POSITION in the `ini` is LinuxCNC build in behaviour. I will check that later today.

My main goal is to get the tool length measured after each tool change. I’m still in the process of understanding how the LinuxCNC configuration works. 

/Mark