kcjengr/probe_basic #8: Question regarding tool touch off

mark0815, 2020-02-16:
Hi.

I'm trying to setup the tool touch off on a switch mounted on the mill table to populate the tool lengths in the tool table.

When doing the tool touch off the values which are written into the tool table do not represent the real (measured with a ruler) tool lengths.

For the setup:
G30 is set up right above the switch with enough clearance. I touched off the spindle nose on the switch and set the G53 Z value in `#<spindle_zero_height>`.

I suspect the these lines in `tool_touch_off.ngc`:

```ngc
#<z_slow_probe> = #5063    (save slow probe result to parameters)
```

and

```ngc
#<new_tool_length_offset> = [ABS[#<spindle_zero_height> + #<z_slow_probe>]]
```

As `#<spindle_zero_height>` is in G53 and `#5063` is in the current WCS the resulting tool length depends on the origin of the WCS.

Do I have a wrong setup or is this an bug?
I want to fix that but I haven't found an equivalent for `#5053` in G53. My solution would be to use another WCS for the tool measurement and set this WCS Z0 to the spindle nose touch off on the tool setter.

KurtJacobson, 2020-02-16:
Hello Mark,

This does indeed look like a bug. I'm not very familiar with @Lcvette 's routines, so will have to wait for him to respond, but I thing if change the line to this it will work correctly:  

`#<z_slow_probe> = #5063  + #[5203 + [20 * #5220]]`

What this does is convert the probed pos to G53 coords by adding the Z offset of the current coordinate system to the probe position.

Lcvette, 2020-02-16:
the tool touch off is very limitedly tested, i only have the simulator to test with at the moment.  all of the tool table and probing is setup to use the spindle nose as the gauge line  for all other tools to be measured against.  this way when the touch probe is used with its tool length offset active it sets the height for all other tools in the tool table so not other measuring is required if the other tools are set in repeating height holders.  please by all means if you have a touch plate and can help test these routines, that would be fantastic.  I have not used a touch plate before so most of what i wrote was of the cuff.

so the tol touch off should be considered a starting point and NOT a completed macro.  requires user intervention at the moment until i have a working touch plate to do rreal testing on like ive done with the touch probe routines..

mark0815, 2020-02-16:
Sure I will help as I want to get it up and running ;-)

I changed `tool_touch_off.ngc` as suggested and the G38.2 result in G53 looks good:

```
#<z_slow_probe> = [#5063 + #<workspace_z>]
```

`#<workspace_z>` is already defined at the top of the file.

To make it work I also had to `ABS[]` `spindle_zero_height` and `z_slow_probe` as on my machine Z0 is the top most location on the Z-axis.

```
#<new_tool_length_offset> = [ABS[#<spindle_zero_height>] - ABS[#<z_slow_probe>]]
```

I added another script to touch off the spindle nose on the tool setter. As I have to home my machine on every startup this is necessary for repeatability.

A second addition I made is a script to set my probe (haimer like) on the tool setter quickly.

Once it's polished I will create a pull request.

Lcvette, 2020-02-16:
> Sure I will help as I want to get it up and running ;-)
>
> I changed `tool_touch_off.ngc` as suggested and the G38.2 result in G53 looks good:
>
> ```
> #<z_slow_probe> = [#5063 + #<workspace_z>]
> ``` [...]

fantastic!! that would be great! and thank you for the help on this one!!

d3cor, 2020-04-04:
Hello,

I can confirm that the changes made the tool touch off work for me too. Thanks :)

I have another question. 

I have no ATC and no SK30 or similar and my tool height will change on every tool change. Can i call the tool touch off macro when a M6 is done?

I also have a bug if i use the "M6 G43" button at the main page. 

Here is the error message:

[qtpyvcp.actions.machine_actions][INFO]  Issuing MDI command: o<m6_tool_call_main_panel> call [3.0] (machine_actions.py:207)
QFileSystemWatcher::removePath: path is empty
QFileSystemWatcher::addPath: path is empty
[qtpyvcp.widgets.input_widgets.gcode_editor][ERROR]  File path is not valid:  (gcode_editor.py:515)
[qtpyvcp.vcp_launcher][CRITICAL]  Unhandled exception in vtk_backplot.py line 397 (vcp_launcher.py:38)
Traceback (most recent call last):
  File "/home/jan/.local/lib/python2.7/site-packages/qtpyvcp/widgets/display_widgets/vtk_backplot/vtk_backplot.py", line 397, in load_program
    self.canon.draw_lines()
  File "/home/jan/.local/lib/python2.7/site-packages/qtpyvcp/widgets/display_widgets/vtk_backplot/vtk_backplot.py", line 220, in draw_lines
    path_actor.poly_data.SetPoints(path_actor.points)
AttributeError: 'NoneType' object has no attribute 'poly_data'

Best regards,
Jan

EderenBarnworks, 2020-04-04:
Hi,

Yes you can include tool touch of in M6. You can give it an own M Code and call it after M6 with the help of your post processor.
Or you can call it from your custom M6.
I use a custom M6 for my machine and it also includes a manual tool change for tools not loaded to the atc. 
So in you case I would write me my own code chis also will bring the spindle to the tool change position and call the touch off afterwards.

You need to remap your existing M6 

Here is a part of my ini:

SUBROUTINE_PATH = ~/linuxcnc/configs/IgorProbe/nc_files/sub_progs/
USER_M_PATH = ~/linuxcnc/configs/IgorProbe/nc_files/
REMAP= M6 modalgroup=6 ngc=toolchange2

A part of my tool chang (shortened and not tested an I added the touch off call):

o<toolchange2> sub
(M108 P#<_current_tool>  Q#<_selected_tool> )

;assign the variables passed by M6 to some parameters
#100 = #<_selected_tool>
#110 = #<_current_tool>
(#120 = #<selected_pocket>)

O10 if [#100 EQ #110] ; checks if tool in the spindle is same as requested
 o<toolchange2> endsub [1]
M2
O10 endif

O200 IF [#100 GT 0]  ; selected tool is not tool0
    	O210 if [#<next_pocket> EQ 0] ;if new tool is not in a pocket
                ; Go for manual Tool load
        	M113		; Raise basket
                G53 G0 Z0       ; Raise spindle
                G53 G0 X0 Y0	; move to door
                M110            ; draw bar loose
                M126            ; manual tool load
                ;M6 G43
	O210 endif
       
O200 Else
        ;new tool is 0 raise basket and clamp drawbar  
        M113 ;raise basket
	M111 ;clamp drawbar
	;M6 G43
O200 ENDIF

M61 Q#100

o<tool_touch_off> call

o<toolchang2> endsub [1]

M2

Since the last update I get the same one but ignored it if it is not gone with the next update I ask how to fix it.

Have a nice weekend Patrick

d3cor, 2020-04-05:
Thanks Patrick!
 
I edited your macro a little bit and now everything works. I copyed the "tool_touch_off" sub because the original is filled with variables from the gui and at the normal M6 gcode the gui is not involved. 

I dont know if this can be done more the "probe basic" way but at the moment it is working like i need it.

Best regards,
Jan

Lcvette, 2021-01-31:
excellent, I will close this issue seems like we are in business and everything working!

closed 2021-01-31 by Lcvette