kcjengr/probe_basic #10: Touch Off Retract Error

DaveRosario, 2020-04-17:
Hi,

I just wanted to report a issue I ran into, just in case someone is having the same issue.

When I touch off a long tool (3.4 in) I get the following errors:

_Linear move on line 54 would exceed Z's positive limit
Linear move on line 54 would exceed joint 2's positive limit
invalid params in linear command_

I went to line 54 in the tool_touch_off.ngc file

and change:
G90
  G0 Z[#<z_slow_probe> + #<retract_distance>]

With:
  G91
  G0 Z0.125

This just retract the z0 1/8 of an inch only and it seems to fix the issue in my machine.

I previously had the same issue mark0815 reported on feb 16, so I modified the code with the suggested modifications in that thread.

My machine is a OMIO CNC X4-800L-USB
max Z travel is 4.3 in

-David

Lcvette, 2020-04-18:
> Hi,
>
> I just wanted to report a issue I ran into, just in case someone is having the same issue.
>
> When I touch off a long tool (3.4 in) I get the following errors:
>
> _Linear move on line 54 would [...]

did you set a retract distance on the tool touch off parameters page? and was the retract distance small enough not to exceed your max z travel?

Lcvette, 2020-04-18:
[image]
the settings might look something like this but you need to measure your spindle zero height for your machine so it knows where the spindle nose gauge line is for the subroutine to work.

Lcvette, 2020-04-18:
```
(Slow Probe Rule, if Slow Probe FR is set to 0, Slow Probe is Bypassed)
o<120> if [#<slow_probe_fr> GT 0]
  (Initiate Slow Z- Probe)
  G91
  F[#<slow_probe_fr>]    (set probe slow feedrate)
  G38.2 Z-[#<retract_distance> * 2]    (slow tool probe)
  #<z_slow_probe> = #5063
  (debug, Probed Pos: #5063 Z Pos: #<_z>)
  G90
  G0 Z[#<z_slow_probe> + #<retract_distance>]
o<120> endif```

this code is showing that the probed position #5063 will have the specified retract distance added to it to make the G90 move.  this gives a move relative to the actual probed position stored in the parameter file so the move is above the touch plate the exact specified distance.

DaveRosario, 2020-04-18:
Thank you guys for the fast response,

I noticed I had SPINDLE ZERO = 0 I changed to 4.3 in and now it seems to be working fine with the original code.
I always had RETRACT DIST = 0.06

Thanks for the GUI, I hope this become Linuxcnc's official GUI! 
Great Job!

Here are my new settings that work:

[image]

Lcvette, 2021-01-31:
welcome!  :D

closed 2021-01-31 by Lcvette