kcjengr/probe_basic #11: Main window displays wrong offset when calling a subroutine from a ngc file

DaveRosario, 2020-04-18:
Hi,

I noticed if I call a subroutine from a .ngc the main window doesn't display the part in G54 0,0,0 but in G53 0,0,0:
[image]

However if I load the same file in Axis it shows it in the right position:
[image]

I don't know if the way I'm calling the subroutine is the best way but it works. The subroutine is to touch off on every tool change.

Thanks,
David

TurBoss, 2020-04-19:
Hello, 

I tried calling a sub but didn't catch the bug
I'm not sure how to reproduce your issue

it looks like there is a bug on the 3d plotter

[image]

DaveRosario, 2020-04-19:
Hi,
Thanks for the super fast response.
base on your screen shot, if I remove the "call"  so: o<touch_off>
it does display the part in the right location but it doesn't execute the subroutine.
I don't know if I'm missing something here.

Thanks,
David

TurBoss, 2020-04-19:
yes my fault

TurBoss, 2020-04-19:
the sub gets called and path is shown in the correct place

o <go_to_g30> call

DaveRosario, 2020-04-19:
You are right,  if I do "o <go_to_g30> call"  it works fine. 
There might be something in the touch_off routine that the plotter doesn't like.
I'll look into it and report back.
Thanks

DaveRosario, 2020-04-19:
So, I wast trying to reuse the tool_touch_off.ngc subroutine but it looks like when it switches to 
G59.3 it shows the part in the origin as G59.3 x,y,z offsets are 0,0,0.

if I comment out G59.3 it shows the part in the G54 offsets but it give me some errors
I couldn't figure out how to reuse this so I'll have to write a subroutine for this.
if you have an Idea how to modify this to make it work right please let me know.

Thanks,
David

o<touch_off> sub

#<fast_probe_fr> = 80    (set from probe screen fast probe feed rate)
#<slow_probe_fr> = 1    (set from probe screen slow probe feedrate)
#<z_max_travel> = 2.8    (max z distance the tool travels before erroring out if not contact is made)
#<xy_max_travel> = 0    (max xy distance the tool travels before erroring out if not contact is made)
#<retract_distance> = 0.06    (distance the tool retracts after making contact during fast feed mode)
#<spindle_zero_height> = 4.3    (G53 distance from home to spindle nose triggering point on touch plate)
#<tool_diameter_probe_mode> = 0    (activates the tool diameter probe subroutine section)
#<tool_diameter_offset_mode> = 0    (activates the tool diameter offset position for probe subroutine section)
#<tool_diameter> = 0   (current tool's diameter used for offseting probe position in x axis)

G92.1    (Cancel G92 offset)

#<workspace_z> = #5220

#<tool_touch_x_coords> = #5181
#<tool_touch_y_coords> = #5182
#<tool_touch_z_coords> = #5183

#<tool_radius_offset> = [#<tool_diameter> / 2]
#<offset_probing_position> = [#<tool_touch_x_coords> - #<tool_radius_offset>]

G59.3

o<100> if [#<tool_diameter_offset_mode> EQ 1]
  #<tool_touch_x_coords> = #<offset_probing_position>
o<100> endif

o<110> if [2 EQ 2]
  G49
o<110> endif

G90    (set absolute coordinates)
G53 G0 Z0    (move to z0 home position)
G53 G0 X#<tool_touch_x_coords> Y#<tool_touch_y_coords>
G53 G0 Z#<tool_touch_z_coords>

G91
F #<fast_probe_fr>
G38.2 Z-[#<z_max_travel>]    (fast tool probe)
#<z_fast_probe> = #5063    (save probe result of fast probe to parameters)
G0 Z[#<retract_distance>]    (retract tool retract distance amount)

(Slow Probe Rule, if Slow Probe FR is set to 0, Slow Probe is Bypassed)
o<120> if [#<slow_probe_fr> GT 0]
  (Initiate Slow Z- Probe)
  G91
  F[#<slow_probe_fr>]    (set probe slow feedrate)
  G38.2 Z-[#<retract_distance> * 2]    (slow tool probe)
  #<z_slow_probe> = [#5063 + #<workspace_z>]
  (debug, Probed Pos: #5063 Z Pos: #<_z>)
  G90
  G0 Z[#<z_slow_probe> + #<retract_distance>]
  (G91)
  (G0 Z0.125)
o<120> endif

o<130> if [#5070 EQ 1]    (verify probe event was succesful)
  #<z_slow_probe> = #5063    (save slow probe result to parameters)
o<130> else
  (MSG,Tool Length Offset Probe Failed)
o<130> endif

(Tool Diameter Probe Mode Section, User must define this section as needed)
o<140> if [#<tool_diameter_probe_mode> EQ 1]
  (MSG, Tool Diameter Probing is Not Defined in Subroutine)
o<140> endif

o<150> if [3 EQ 3]
  G49
o<150> endif

G90    (set absolute coordinates)
G53 G0 Z0 (Send Spindle to home zero position)

(define new tool length offset parameters)
#<new_tool_length_offset> = [ABS[#<spindle_zero_height>] - ABS[#<z_slow_probe>]]

G10 L1 P #5400 Z [#<new_tool_length_offset>]  (5400 = tool number)

o<160> if [#<workspace_z> EQ 1]
  G54
o<160> else if [#<workspace_z> EQ 2]
  G55
o<160> else if [#<workspace_z> EQ 3]
  G56
o<160> else if [#<workspace_z> EQ 4]
  G57
o<160> else if [#<workspace_z> EQ 5]
  G58
o<160> else if [#<workspace_z> EQ 6]
  G59
o<160> else if [#<workspace_z> EQ 7]
  G59.1
o<160> else if [#<workspace_z> EQ 8]
  G59.2
o<160> endif

T #5400 G43  H #5400    (enable tool length offset)

o<touch_off> endsub

M2

DaveRosario, 2020-04-25:
Hi, 

So I think I figured it out.

Instead of switching to G59.3 I'm using G28.1 to get the absolute position of the z axis of the slow probe. It works pretty good so far.

Here is the whole modified code:

o<touch_off> sub

#<fast_probe_fr> = 80    (set from probe screen fast probe feed rate)
#<slow_probe_fr> = 1    (set from probe screen slow probe feedrate)
#<z_max_travel> = 3.5    (max z distance the tool travels before erroring out if not contact is made)
#<xy_max_travel> = 0    (max xy distance the tool travels before erroring out if not contact is made)
#<retract_distance> = 0.06    (distance the tool retracts after making contact during fast feed mode)
#<spindle_zero_height> = 3.85    (G53 distance from home to spindle nose triggering point on touch plate)
#<tool_diameter_probe_mode> = 0    (activates the tool diameter probe subroutine section)
#<tool_diameter_offset_mode> = 0    (activates the tool diameter offset position for probe subroutine section)
#<tool_diameter> = 0   (current tool's diameter used for offseting probe position in x axis)

G92.1    (Cancel G92 offset)

#<workspace_z> = #5220

#<tool_touch_x_coords> = #5181
#<tool_touch_y_coords> = #5182
#<tool_touch_z_coords> = #5183

#<tool_radius_offset> = [#<tool_diameter> / 2]
#<offset_probing_position> = [#<tool_touch_x_coords> - #<tool_radius_offset>]

(G59.3)

o<100> if [#<tool_diameter_offset_mode> EQ 1]
  #<tool_touch_x_coords> = #<offset_probing_position>
o<100> endif

o<110> if [2 EQ 2]
  G49
o<110> endif

G90    (set absolute coordinates)
G53 G0 Z0    (move to z0 home position)
G53 G0 X#<tool_touch_x_coords> Y#<tool_touch_y_coords>
G53 G0 Z#<tool_touch_z_coords>

G91
F #<fast_probe_fr>
G38.2 Z-[#<z_max_travel>]    (fast tool probe)
G28.1
#<z_fast_probe> = #5163    (save probe result of fast probe to parameters)
G0 Z[#<retract_distance>]    (retract tool retract distance amount)

(Slow Probe Rule, if Slow Probe FR is set to 0, Slow Probe is Bypassed)
o<120> if [#<slow_probe_fr> GT 0]
  (Initiate Slow Z- Probe)
  G91
  F[#<slow_probe_fr>]    (set probe slow feedrate)
  G38.2 Z-[#<retract_distance> * 2]    (slow tool probe)
  G28.1
  #<z_slow_probe> = [#5163 + #<workspace_z>]
  (debug, Probed Pos: #5163 Z Pos: #<_z>)
  G90
  G0 Z[ABS[#<z_slow_probe> + #<retract_distance>]]
o<120> endif

o<130> if [#5070 EQ 1]    (verify probe event was succesful)
  #<z_slow_probe> = #5163    (save slow probe result to parameters)
o<130> else
  (MSG,Tool Length Offset Probe Failed)
o<130> endif

(Tool Diameter Probe Mode Section, User must define this section as needed)
o<140> if [#<tool_diameter_probe_mode> EQ 1]
  (MSG, Tool Diameter Probing is Not Defined in Subroutine)
o<140> endif

o<150> if [3 EQ 3]
  G49
o<150> endif

G90    (set absolute coordinates)
G53 G0 Z0 (Send Spindle to home zero position)

(define new tool length offset parameters)
#<new_tool_length_offset> = [ABS[#<spindle_zero_height>] - ABS[#<z_slow_probe>]]

G10 L1 P #5400 Z [#<new_tool_length_offset>]  (5400 = tool number)

o<180> if [#<workspace_z> EQ 1]
  G54
o<180> else if [#<workspace_z> EQ 2]
  G55
o<180> else if [#<workspace_z> EQ 3]
  G56
o<180> else if [#<workspace_z> EQ 4]
  G57
o<180> else if [#<workspace_z> EQ 5]
  G58
o<180> else if [#<workspace_z> EQ 6]
  G59
o<180> else if [#<workspace_z> EQ 7]
  G59.1
o<180> else if [#<workspace_z> EQ 8]
  G59.2
o<180> endif

T #5400 G43  H #5400    (enable tool length offset)

o<touch_off> endsub

M2

TurBoss, 2020-04-25:
great you found a solution

so I'm closing this, feel free to open any issues you need

closed 2020-04-25 by TurBoss