kcjengr/probe_basic #13: feed rate override reset

lmclaren7000, 2020-05-24:
I would like to be able to reset the feed rate after it has been overridden. The UI has a button "F 100%" and I would like to be able to call the same action via a pin. The UI button does not appear to be available from halui.

Can the same also be done for the Spindle as well.

An example from a different UI is shown here:
(https://forum.linuxcnc.org/49-basic-configuration/28974-hardware-button-to-reset-spindle-feed-override)

lmclaren7000, 2020-05-25:
further information:
If  you use the UI slider or an MPG to change the feedrate-override the slider moves in sync.
If you want to reset the feedrate-override then the F 100% button will set the halui.feed-override.value (Output) back to 1.

If you are using a MPG to change the feedrate-override the count of the MPG is not related to zero, eg it could read 543 for 100% feedrate-override, it seems to take whatever the count is when started and then increment or decrement the feedrate-override based on the difference not the absolute value, this means you can not reset the MPG count either.

I would like to connect a pin to a signal that will reset the feedrate to the same value as the UI button (F 100%) but there does not appear to be any input that would permit this.

joco-nz, 2020-05-26:
This pull request at the framework level should solve if approved.
https://github.com/kcjengr/qtpyvcp/pull/37

joco-nz, 2020-06-12:
Pull request #38 on the framework covers this functionality.  This has now been merged.
You do need to ensure the required plugin is activated in the VCP yaml config file.  See the default yaml file in the framework for the required plugin entry for the hal pins export.

Cheers.

closed 2020-11-15 by TurBoss