kcjengr/probe_basic #14: Quick install process issue...

RDTymko, 2020-05-30:
Hi,  I am following all your steps as outlined in the Probe Basic Quick Start Install.  I get to the point where you change the permissions on the ProbeBasic-Installer-v0.2.8.  I make the appropriate changes as directed.  After said changes, I double click the file and the installer does not run, although another file pops into the dir and nothing else happens... dead in the water.  The file that is created is... "XTB@��@8 (invalid encoding)"

Any ideas?

I have tried the developer install to no avail either.

Richard

TurBoss, 2020-05-30:
hello,
what OS are you running?
its only tested on debian 10

TurBoss, 2020-05-30:
hello,

 I found the issue, you need to rename the installer to 

ProbeBasic-Installer-v0.2.8

to 

ProbeBasic-Installer-v0.2.8.run

./ProbeNa

RDTymko, 2020-05-31:
Hi, I’m using the exact code and links that are provided in your instructional page. Debian 9 iso is what your page is linked to. 

I will try renaming the file tomorrow. Thanks for the quick reply. 

Great work by the way. 

Richard
From My iPhone

RDTymko, 2020-05-31:
Hi,  just tried renaming the file and got the same result. Downloaded the file again, renamed it to ProbeBasic-Installer-v0.2.8.run and same result. 

Richard
From My iPhone

TurBoss, 2020-06-02:
hello,

I don't know what can be the issue :/

RDTymko, 2020-06-02:
Hmm.  Do you have a more current installer or an earlier one?

I am following this verbatim including all the download files as per the links… https://kcjengr.github.io/probe_basic/dev_install.html <https://kcjengr.github.io/probe_basic/dev_install.html>

Can you outline an install process without the installer?

Thanks.

TurBoss, 2020-06-02:
there is a way to install probe basic by hand and skip the installer

check this develop installation

https://kcjengr.github.io/probe_basic/dev_install.html

thanks

joco-nz, 2020-06-02:
I am not convinced the .run extension is needed.  Unless that is a quirk of the OS you are running.  I just tried this under LinuxMint and per the instructions all worked.
As a suggestion can try this from the command line after navigating using the 'cd' command to the directory where your PB installer is located.

chmod +x ProbeBasic-Installer-v0.2.8
sudo ./ProbeBasic-Installer-v0.2.8 

See what happens and let us know any text that gets send to the command terminal.

Thanks.

TurBoss, 2020-06-02:
is the installer in a directory named "probe_basic"
like
```bash
~/probe_basic/ProbeBasic-Installer-v0.2.8
```
if yes move it to another place :D

RDTymko, 2020-06-02:
Moved the installer to linuxcnc.
Tried the process with and without the .run
Through terminal i did the following...
rdtymko@LinuxCNC:~$ ls 
Desktop    Downloads  Music     Public     Videos
Documents  linuxcnc   Pictures  Templates
rdtymko@LinuxCNC:~$ cd linuxcnc
rdtymko@LinuxCNC:~/linuxcnc$ ls
configs  nc_files  ProbeBasic-Installer-v0.2.8.run
rdtymko@LinuxCNC:~/linuxcnc$ ls
configs  nc_files  ProbeBasic-Installer-v0.2.8
rdtymko@LinuxCNC:~/linuxcnc$ chmod +x ProbeBasic-Installer-v0.2.8.run 
rdtymko@LinuxCNC:~/linuxcnc$ sudo ./ProbeBasic-Installer-v0.2.8.run
[sudo] password for rdtymko: 
./ProbeBasic-Installer-v0.2.8: 1: ./ProbeBasic-Installer-v0.2.8: ELF: not found
./ProbeBasic-Installer-v0.2.8: 2: ./ProbeBasic-Installer-v0.2.8: Syntax error: ")" unexpected

Then same result without the .run...

rdtymko@LinuxCNC:~$ ls 
Desktop    Downloads  Music     Public     Videos
Documents  linuxcnc   Pictures  Templates
rdtymko@LinuxCNC:~$ cd linuxcnc
rdtymko@LinuxCNC:~/linuxcnc$ ls
configs  nc_files  ProbeBasic-Installer-v0.2.8
rdtymko@LinuxCNC:~/linuxcnc$ ls
configs  nc_files  ProbeBasic-Installer-v0.2.8
rdtymko@LinuxCNC:~/linuxcnc$ chmod +x ProbeBasic-Installer-v0.2.8 
rdtymko@LinuxCNC:~/linuxcnc$ sudo ./ProbeBasic-Installer-v0.2.8
[sudo] password for rdtymko: 
./ProbeBasic-Installer-v0.2.8: 1: ./ProbeBasic-Installer-v0.2.8: ELF: not found
./ProbeBasic-Installer-v0.2.8: 2: ./ProbeBasic-Installer-v0.2.8: Syntax error: ")" unexpected

When you tried under linuxmint was it installed on HD?

Thanks

TurBoss, 2020-06-02:
I'm working on the installer stuff it will take a day or two

RDTymko, 2020-06-02:
NP.  Wondering if it could have something to do between 32bit 64 bit?
just thinking out loud?

TurBoss, 2020-06-02:
32 bit OS is not supported by the installer and some other parts likee the 3d plotter
so it would not work on 32 :(

RDTymko, 2020-06-04:
Ok,  After some sleuthing and trying Linux mint as joco-nz mentioned,  I got it up and installed.  The problem was that I was using the 32bit ISO which I did not realize I selected.  So... One problem down.  

So how do I get it to work on my cnc?  I can run the stepconfig for a parallel cable configuartion  and get the default interface working and operating my machine,  but when I launch Probe_Basic it boots up but does not control my machine.  How do I setup Probe basic to work on my machine?

I cannot find an instructional anywhere.

BTW...  I have never used linuxcnc before or linux for that matter.

Thanks for your help.  It is a great interface! #

TurBoss, 2020-06-04:
stepconf will generate a INI file you only need to rename the DISPLAY option to use probe_basic instead of axis

closed 2020-11-15 by TurBoss