kcjengr/probe_basic #18: Program and machine labels drawn twice

chetmurphy, 2020-07-16:
I'm seeing all axis labels being drawn twice. All other graphics look OK and on axis and gmocrapy the labels are drawn correctly.

I'm using the latest installer on a new machine running Linux Mint updated. I've tested with both the on-board graphics and an older NVIDIA card with the same results.

TurBoss, 2020-07-25:
Hello, could you pleas attach a screenshot?

sorry for the delay

chetmurphy, 2020-07-26:
Thanks for getting back to me. It turns out that the duplicate labels were for the top and bottom of the bounding box. So it looks like the labels are being drawn twice and on top of each other but that depends upon the perspective.

It would be nice if we could turn off the labels for both the top and bottom when the bounding box is not very high.

--Chet

TurBoss, 2020-07-26:
yes is posible to turn them on 

you only need to create a section in your ini that looks like this
https://github.com/kcjengr/qtpyvcp/blob/master/linuxcnc/configs/sim.qtpyvcp/xyz.ini#L57
```
[VTK]
# VTK_Widget Options
MACHINE_BOUNDRY = False
MACHINE_TICKS = False
MACHINE_LABELS = False

PROGRAM_BOUNDRY = False
PROGRAM_TICKS = False
PROGRAM_LABELS = False
GRID_LINES = False

```

TurBoss, 2020-07-29:
@chetmurphy 
Hello,

did you find a solution for this? do the 3d plotter look better??
I think this issue is fixed since new versions of qtpyvcp will have new defaults like hidden labels

I'm closing this, please if you need something feel free to open new issues or reopen :)
Thanks!

closed 2020-07-29 by TurBoss