kcjengr/probe_basic #22: Development Install Description is not up to date?

Gnevko, 2020-11-01:
Hi,
the following command in the Development Install (https://kcjengr.github.io/probe_basic/dev_install.html):

 `cp -r ~/probe_basic/config/xyzab/ ~/linuxcnc/configs/`

cann't be executed:

`cp: cannot stat '/home/cnc/probe_basic/config/xyzab/': No such file or directory`

TurBoss, 2020-11-01:
thanks, please check the up to date docs 
https://www.qtpyvcp.com/install/dev_install.html

Gnevko, 2020-11-01:
sorry for my English, but we speaking about ProbeBasic "Development Install", not about  QtPyVCP ...

TurBoss, 2020-11-01:
oh very true

TurBoss, 2020-11-01:
https://kcjengr.github.io/probe_basic/dev_install.html

qcompile and config path where fixed

it should take some minutes to show up
thanks!

Gnevko, 2020-11-02:
1) config path where fixed - ok
2) qcompile is still missing for **Probe Basic** part - not ok

TurBoss, 2020-11-02:
now its fixed

thanks !

Gnevko, 2020-11-02:
It seams to be ok now, thank you!

closed 2020-11-02 by Gnevko