kcjengr/probe_basic #23: ImportError: No module named probe_basic_rc

Gnevko, 2020-11-01:
just followed development install and then trying to start it with linuxcnc

`Traceback (most recent call last):
  File "/home/cnc/.local/bin/probe_basic", line 11, in <module>
    load_entry_point('probe-basic', 'gui_scripts', 'probe_basic')()
  File "/home/cnc/probe_basic/probe_basic/__init__.py", line 20, in main
    qtpyvcp.run_vcp(opts, VCP_CONFIG_FILE)
  File "/home/cnc/qtpyvcp/qtpyvcp/__init__.py", line 27, in run_vcp
    run(*args, **kwargs)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/__init__.py", line 129, in run
    launch_application(opts, config)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 78, in launch_application
    loadWindows(config['windows'])
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 228, in loadWindows
    window = _initialize_object_from_dict(window_dict)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 203, in _initialize_object_from_dict
    obj = _get_object_by_referance(provider)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 191, in _get_object_by_referance
    return getattr(importlib.import_module(modname), attrname)
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/home/cnc/probe_basic/probe_basic/probe_basic.py", line 12, in <module>
    import probe_basic_rc
ImportError: No module named probe_basic_rc`

TurBoss, 2020-11-01:
hello,

you may need to run the qcompile program under the probe_basic sources directory

```bash
qcompile .
```

Gnevko, 2020-11-01:
can I ask you to write this information into "Development Install" for Probe Basic?

TurBoss, 2020-11-01:
done

Gnevko, 2020-11-02:
qcompile is still missing in the description for **Probe Basic** part - not ok

Gnevko, 2020-11-02:
ok. gcompile is in the place but after the following of all steps, I tied to start probe basic in lunuxcnc with following result:

`Traceback (most recent call last):
  File "/home/cnc/.local/bin/probe_basic", line 11, in <module>
    load_entry_point('probe-basic', 'gui_scripts', 'probe_basic')()
  File "/home/cnc/probe_basic/probe_basic/__init__.py", line 20, in main
    qtpyvcp.run_vcp(opts, VCP_CONFIG_FILE)
  File "/home/cnc/qtpyvcp/qtpyvcp/__init__.py", line 27, in run_vcp
    run(*args, **kwargs)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/__init__.py", line 129, in run
    launch_application(opts, config)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 78, in launch_application
    loadWindows(config['windows'])
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 228, in loadWindows
    window = _initialize_object_from_dict(window_dict)
  File "/home/cnc/qtpyvcp/qtpyvcp/app/launcher.py", line 208, in _initialize_object_from_dict
    return obj(*args, **kwargs)
  File "/home/cnc/probe_basic/probe_basic/probe_basic.py", line 23, in __init__
    super(ProbeBasic, self).__init__(*args, **kwargs)
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/form_widgets/main_window.py", line 43, in __init__
    self.loadUi(ui_file)
  File "/home/cnc/qtpyvcp/qtpyvcp/widgets/form_widgets/main_window.py", line 102, in loadUi
    uic.loadUi(ui_file, self)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/__init__.py", line 226, in loadUi
    return DynamicUILoader(package).loadUi(uifile, baseinstance, resource_suffix)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/Loader/loader.py", line 72, in loadUi
    return self.parse(filename, resource_suffix, basedir)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 1021, in parse
    actor(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 828, in createUserInterface
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 273, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 503, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 546, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 503, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 546, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 503, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 546, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 273, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 273, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 503, in createLayout
[... 11 lines cut ...]
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 273, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 503, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 546, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 806, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 264, in createWidget
    self.stack.push(self.setupObject(widget_class, parent, elem))
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/uiparser.py", line 228, in setupObject
    obj = self.factory.createQObject(clsname, name, args, is_attribute)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/objcreator.py", line 100, in createQObject
    factory = self.findQObjectType(classname)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/objcreator.py", line 125, in findQObjectType
    w = module.search(classname)
  File "/usr/lib/python2.7/dist-packages/PyQt5/uic/Loader/qobjectcreator.py", line 113, in search
    module = __import__(mname, {}, {}, (cls,))
ImportError: No module named qtpyvcp_conversational_gcode.widgets.facing`

What is interesting, probe basic lathe works ok.

TurBoss, 2020-11-02:
oh you need to install gcode_conversational too
https://github.com/kcjengr/qtpyvcp.conversational-gcode
this is new

Gnevko, 2020-11-02:
as usual pls add it to install description :)

TurBoss, 2020-11-02:
done

Thanks!

Gnevko, 2020-11-02:
bevor I will close this issue, just small question - does it make sense to report **probe basic lathe** bugs and not working features or is it to early for that?

TurBoss, 2020-11-02:
>does it make sense to report probe basic lathe bugs and not working features or is it to early for that?

absolutely make any issue you need 
I'll try to help  :-)

Gnevko, 2020-11-02:
Ok :) thank you for your time and work!

closed 2020-11-02 by Gnevko