kcjengr/probe_basic #27: Probe Basic Lathe: documentation

Gnevko, 2020-11-03:
All type of documentation are missing:
1) How to install
2) How to configure
3) How to use

For people who coms from "Mill World" it's not really easy to understand, how a Cnc lathe should be configured and how to use for example tool touch and so on.

Lcvette, 2021-01-31:
this is true!!  you seem very active on this, would you mind maybe assisting in coming up with some documentation?

Lcvette, 2021-01-31:
> All type of documentation are missing:
> 
> 1. How to install
> 2. How to configure
> 3. How to use
> 
> For people who coms from "Mill World" it's not really easy to understand, how a Cnc lathe should be configured and how to use for example tool touch and so on.

this is true!! you seem very active on this, would you mind maybe assisting in coming up with some documentation?