kcjengr/probe_basic #29: Probe Basic Lathe: delete tool and save table

Gnevko, 2020-11-03:
1) Set tool 1 as current tool ( on the main or touch off screen)
2) delete tool 1 from the tool table (on the tool table screen)
3) Press "Save table" button

Traceback (most recent call last):
  File "/home/cnc/qtpyvcp/qtpyvcp/plugins/tool_table.py", line 250, in reloadToolTable
    tool_table = self.loadToolTable()
  File "/home/cnc/qtpyvcp/qtpyvcp/plugins/tool_table.py", line 324, in loadToolTable
    self.current_tool.setValue(self.TOOL_TABLE[STATUS.tool_in_spindle.getValue()])
KeyError: 1

Lcvette, 2021-01-31:
I am able to reproduce error and will look at it!

KurtJacobson, 2021-01-31:
A warning dialog will now be shown when trying to delete a tool that is currently in the spindle.

[image]

KurtJacobson, 2021-01-31:
I guess the message should read different for a lathe ...

closed 2021-01-31 by KurtJacobson