kcjengr/probe_basic #36: Probe Basic Lathe: orientation of tool path

Gnevko, 2020-11-09:
With the same configuration in Axis:

[image]

Probe Basic Lathe:

[image]

TurBoss, 2020-11-13:
the view is inverted I'll take a look

TurBoss, 2020-11-14:
it has to do with the default view

can you try to change view from the "View" menu then "View Angle"and "XZ Lathe" or "XZ2
Lathe"

Thanks

Gnevko, 2020-11-14:
Thank you! 
1) It works, but I would suggest to use BACK_TOOL_LATHE = 1 parameter from INI File, to manage the view orientation. 

2) "View" menu as a lot of other items, but I'm not sure if all of them make sense  for  the Lathe
[image]

3) The view displays only machine coordinate system
[image]

I didn't  found any possibility to change it for work coordinates. It seams, that the following menu Items don't work correctly
[image]

TurBoss, 2020-11-15:
makes sense to hide non lathe options when lathe is setup and the opposite

machine and part bounds require a bit of work :|

closed 2021-05-11 by Lcvette