kcjengr/probe_basic #37: Critical Error: Tool N can't be found

Gnevko, 2020-11-11:
LinuxCnc 2.9
ProbeBasic - development

Any changes in the tool table leads to the following error: linuxcnc cann't "see" all tools anymore. If I try to use for example "T1 M6"  command then I receive the error message: "Tool 1 can't be found". After that it helps only restore tool file from my backup.

TurBoss, 2020-11-11:
Just tested on 2.80 and save seems to work
🤔

TurBoss, 2020-11-11:
Hello,

currently probe_basic develop install install latest qtpyvcp but not git branch "master"
could you please if not already install qypyvcp in develop mode 

it should be already fixed

Thank you!

Gnevko, 2020-11-13:
Hi, I had already development version installed. Today I reinstall it again and it looks fine now.
Thank you.
(PS: about PB Lathe - is the live here?)

TurBoss, 2020-11-13:
Hello,

Thanks

about lathe I'll try to address the issues

closed 2020-11-13 by TurBoss

Gnevko, 2020-11-14:
Hm ...  now I have the same issue with Probe Basic Lathe. Any changes lead to corrupted tool table file :(

Lcvette, 2021-01-31:
fixed