kcjengr/probe_basic #40: Probe Basic Lathe: Critical Error: Tool N can't be found

Gnevko, 2020-11-14:
It's a bit strange, but the loading of latest development version of qtpyvcp fix the problem within Probe Basic, but now I have the same issue within Probe Basic lathe!

What it interesting, with previous dev version of qtpyvcp the tool table in the Probe Basic lathe had work correctly.

Lcvette, 2021-01-31:
can you elaborate on the issue im trying to reproduce in both release and dev version

Lcvette, 2021-01-31:
I tried loading all tools and it is working correctly as far as i can see and loading a tool not defined in the tool table results in an error message. marking closed, please reopen with more information!  Thanks Gnevko!

closed 2021-01-31 by Lcvette

Lcvette, 2021-01-31:
think we found the problem and have corrected it will be fixed in next release