kcjengr/probe_basic #43: ToolTable gets Blank on Chars like Ä Ü Ö

Moshkopp, 2021-03-09:
Hey there, 

got an an issue on save the toolTable, if I use  Ä Ö Ü in  comment field. 

`Traceback (most recent call last):
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 252, in saveToolTable
    self.tool_model.saveToolTable()
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/widgets/input_widgets/tool_table.py", line 200, in saveToolTable
    self.tt.saveToolTable(self._tool_table, self._columns)
  File "/home/cnc/.local/lib/python2.7/site-packages/qtpyvcp/plugins/tool_table.py", line 422, in saveToolTable
    fh.write('\n'.join(lines))
UnicodeEncodeError: 'ascii' codec can't encode character u'\xe4' in position 97: ordinal not in range(128)`

after that Msg, my ToolTable gets complete empty.

TurBoss, 2021-03-10:
Hello,

this is a known issue that may be fixed

what version do you use?

thanks

find latest version in the release section

Moshkopp, 2021-03-11:
Hello,

I installed the latest stable, an also the develop from Masterbranch. 

`cnc@cnc:~$ probe_basic -v
QtPyVCP 0.3.13+1.g2a45555e.3.gd45a9315, Probe Basic vv0.3.4+5.g4623efa.dirty
cnc@cnc:~$ ^C
`

TurBoss, 2021-03-11:
thank you so much I thin I managed to fix 

if you could test latest qtpyvcp from git?

TurBoss, 2021-03-11:
> this is a known issue that may be fixed

thatwas other issue actually

Moshkopp, 2021-03-11:
Runs like a charme, you fixed it. Thank you.

TurBoss, 2021-03-11:
Thank you too!

closed 2021-03-11 by TurBoss