kcjengr/probe_basic #48: ATC screen blank

aaroncnc, 2021-04-05:
Hello,
I am getting errors when selecting the ATC screen.
I am running on a RPI4 with scottalford75's VTK file 

> https://github.com/scottalford75/LinuxCNC-on-RPi/blob/master/Pi%20OS/vtk_7.1.1-qt5py2-1_armhf.deb
sudo dpkg -i vtk_7.1.1-qt5py2-1_armhf.deb

This is the error seen in terminal

> pi@linuxcnc:~ $ linuxcnc
LINUXCNC - 2.8.1
Machine configuration directory is '/home/pi/linuxcnc/configs/probe_basic'
Machine configuration file is 'probe_basic.ini'
Starting LinuxCNC...
File: tool.tbl Unrecognized line skipped: ;Tool  Pocket Z Offset     Diameter     Remark
twopass:invoked with <> options
twopass:found ./hallib/core_sim_5.hal
twopass:found ./hallib/spindle_sim.hal
twopass:found ./hallib/simulated_home_probe_basic.hal
twopass:found ./hallib/sim_cannon.hal
Note: Using POSIX realtime
motion.c: Creating unlock hal pins for joint 4
qt5ct: using qt5ct plugin
qt5ct: D-Bus system tray: no
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
qt5ct: D-Bus global menu: no
qt5ct: custom style sheet is disabled
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
QEGLPlatformContext: eglMakeCurrent failed: 3002
QQuickWidget: Failed to make context current
QQuickWidget::resizeEvent() no OpenGL context
QEGLPlatformContext: eglMakeCurrent failed: 3002
QQuickWidget: Failed to make context current
QQuickWidget: Attempted to render scene with no context
QEGLPlatformContext: Failed to make temporary surface current, format not updated (3002)
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed
QEGLPlatformContext: eglMakeCurrent failed: 3002
composeAndFlush: makeCurrent() failed

aaroncnc, 2021-05-11:
I have not found a solution as of yet, shelved that part of the project while i focus on the hardware control side.
I looked around for solutions to related errors but came up empty besides the different versions of open gl that the rpi4 supports vs what this supports.

aaroncnc, 2021-11-03:
I do have a slight update.
I switched to a 64bit version of my OS and remade the VTK to support it.
ATC screen is no longer blank.
However.
In the conversational tab the following are blank
Perimeter + all tabs
Pockets + all tabs
Misc + all tabs
Holes: Sport drill, Ream, chamfer, Ridgid Tap, Thread mill, Generate G code

Is there a way to run with more logs so i can see whats going on?
the only hint the terminal gives me is
qt5ct: custom style sheet is disabled
libpng warning: iccp incorrect sRGB profile.

aaroncnc, 2021-11-04:
another update. still not fixed but i got the following to not show in terminal
qt5ct: custom style sheet is disabled
libpng warning: iccp incorrect sRGB profile.

finds out i had to edit a file on my os
 edit (as sudo) /etc/X11/Xsession.d/99qt5ct

Edit 99qt5ct to match below. 
QT_QPA_PLATFORMTHEME=gtk2

Now in terminal it says it loaded the probe_basic.qss

but the pages are still blank and there are no more warnings or info in terminal.

hardware-crash, 2021-11-04:
@aaroncnc 
The blank sub-tabs in Conversational tab are right.
They simply are blank at the moment as there is undergoing development fot these features.
The warning of incorrect sRGB profile is just a warning and shouldnt be a problem.

aaroncnc, 2021-11-04:
@hardware-crash 
Okay well then it seems my install is fully functional now.
if anyone else has some trouble i wrote up most of my steps.
https://forum.linuxcnc.org/qtpyvcp/44120-instructions-for-qtpyvcp-for-python-2-7-to-work-on-rpi-4-64bit

Maybe disable the buttons for not implemented features/screens?
Or just a note of not implemented or comment somewhere.

closed 2021-11-04 by aaroncnc