kcjengr/probe_basic #49: Spindle inhibit while probing

pommen, 2021-04-05:
Hi!
Is there anyway to stop the spindle from starting to rotate while T99 is loaded in the spindle?
Just had a near-expensive accident. =P

What I did wrong:
Had T99(my touch probe) loaded in the spindle, and pressed "cycle start" on a loaded program. E-stop saved me, but thought I would share my experience.

/Peter

KurtJacobson, 2021-04-25:
This is an excellent idea. Probably something that should be done in HAL.

Lcvette, 2021-05-11:
this is a good idea!

TooTall18T, 2022-01-16:
You can write a M300/M400 command where you can check if there is a specific tool loaded. If not the spindle can start.
This is only to use in nc code. M300 instead of M3. The buttons will still start the spindle.

This example is not tested on a real machine!
For example (m300.ngc in subroutines folder):
o<300> sub

;Check if the current tool is T99
o<10> if [#<_current_tool> NE 99]
;If not start the spindle
M3
;(debug, is not 99)
o<10> else
;If it is T99 stop program
(debug, is 99)
M30

o<10> endif

o<m300> endsub

M2 (end program)

The remap line to put into the .ini file:
[RS274NGC] <- put the line below at the end of this section
REMAP=M300 modalgroup=7 ngc=m300

Lcvette, 2023-02-11:
this has been addressed in the latest python3 version

closed 2023-02-11 by Lcvette