kcjengr/probe_basic #63: LinuxCNC -grblHAL

eecue, 2022-05-26:


KurtJacobson, 2022-05-26:
One of the intents of QtPyVCP (the framework used to build the ProbeBasic UI) is for it to be modular enough that it could be used with a "CNC control" other than LinuxCNC (I had GRBL in mind). The goal would be to abstract the interface with the control and make it pluggable. There has been a lot of work in this direction, and most of the communication with LinuxCNC is thru plugins, the main exception being the Actions. The problem is that there is still a good bit of code scattered about that relies on LinuxCNC being present.

I would start by unloading any LinuxCNC specific plugins (comment out in YAML file), and removing any references to linuxcnc and getting a bare GUI working, and then add back what is needed to communicate with grblHAL.

That is, unless you want to try to completely abstract the control interface.. This would be awesome, but a huge amount of work.

eecue, 2022-05-26:
Thanks! Mind if I leave this open for further questions? I realize it’s not actually an issue

eecue, 2022-05-27:
Closing this. Will make a new ticket to discuss things in the QtPyVCP

closed 2022-05-27 by eecue