kcjengr/probe_basic #65: intall issues

chienMouille, 2022-09-21:
Hi, I'm having an installation issue with non-dev install. It says this:

Error during installation process (com.probebasic.core):
Execution failed (Unexpected exit code: 1): "python -m pip install --user --upgrade git+https://github.com/kcjengr/qtpyvcp.git@python2_maintenance"

If I run command in terminal details are as such:

Collecting Jinja2<4,>3 (from HiYaPyCo->qtpyvcp==0.3.19+17.g038cece1)
 Could not find a version that satisfies the requirement Jinja2<4,>3 (from HiYaPyCo->qtpyvcp==0.3.19+17.g038cece1) (from versions: 2.0rc1, 2.0, 2.1, 2.1.1, 2.2, 2.2.1, 2.3, 2.3.1, 2.4, 2.4.1, 2.5, 2.5.1, 2.5.2, 2.5.3, 2.5.4, 2.5.5, 2.6, 2.7, 2.7.1, 2.7.2, 2.7.3, 2.8, 2.8.1, 2.9, 2.9.1, 2.9.2, 2.9.3, 2.9.4, 2.9.5, 2.9.6, 2.10, 2.10.1, 2.10.2, 2.10.3, 2.11.0, 2.11.1, 2.11.2, 2.11.3)
No matching distribution found for Jinja2<4,>3 (from HiYaPyCo->qtpyvcp==0.3.19+17.g038cece1)

Any chance you can help? 

thanks a lot!

TurBoss, 2022-09-22:
Hello,

just fixed by setting HiYaPyCo version to 0.5.0

could you try again plz?

thanks

chienMouille, 2022-09-22:
This solved it as well as the subsequent errors I was getting if ignoring this one.
All running now!
Thanks a lot!

TurBoss, 2022-09-22:
Than you!

closed 2022-09-22 by TurBoss