kcjengr/probe_basic #67: can't download install script from quick start page

piflixe, 2022-12-03:
in the [quick start guide](https://kcjengr.github.io/probe_basic/quick_start.html) the download of an [installer](https://repository.qtpyvcp.com/repo/pb-py2/bin/ProbeBasic-Installer-py2+3.g231c7ff.run) is recommended. 

The link seems not to work. Or am I just doing something very wrong? Tried klicking on the link and also using wget from the command line so far.

Andreg3127, 2022-12-07:
I am also having the same issue. I cannot download the installer script through my CNC computer or through my windows PC. Link appears to be dead.

TurBoss, 2022-12-07:
hello, the server hosting that is offline

will be up as soon as possible

I'll report here

Thanks

TurBoss, 2022-12-16:
and fixed
we are online again!

closed 2022-12-16 by TurBoss