kcjengr/probe_basic #69: Install incomplete (qtpyvcp_conversational_gcode-main)

ferrumd, 2023-03-20:
when trying to install it fails whilst installing "qtpyvcp_conversational_gcode-main"
`Traceback (most recent call last):
  File "./setup.py", line 1, in <module>
    import versioneer
  File "/home/dgentile/Downloads/qtpyvcp_conversational_gcode-main/versioneer.py", line 418
    LONG_VERSION_PY: Dict[str, str] = {}
                   ^
SyntaxError: invalid syntax
`

hasechris, 2023-04-01:
just want to second this. Sadly now im stuck with a non-functioning cnc machine.

TurBoss, 2023-04-02:
conversational gcode got  integradted into probe basic

no need to install it anymore

Alexfehr1987, 2023-04-25:
same issue for me. old 2.8.4 didn't boot anymore, tried debian 12 but it was totally instable on my machine. then tried to install "old"-working 2.8.4 but probebasic is not working there anymore because of conversational error....now i have a nonworking cnc

Lcvette, 2026-02-12:
This version is no longer available or supported, newer python3 version are resolved

closed 2026-02-12 by Lcvette