kcjengr/probe_basic #70: Update not possible

hasechris, 2023-04-01:
Hi,

i wanted to update my probe_basic installation. Sadly i get the following error:

```shell
linuxcnc@linuxcnc:~$ ./probe_basic_installation/PBMaintenanceTool 
[2930] Warning: Network error while downloading 'https://repository.qtpyvcp.com/repo/pb/repo/Updates.xml?39005568': Error transferring https://repository.qtpyvcp.com/repo/pb/repo/Updates.xml?39005568 - server replied: Not Found.
[2931] Warning: Kann entfernten Baum nicht empfangen Keine Paketquellen für diese Anwendung festgelegt.
```

TurBoss, 2023-04-02:
Hello, sorry for the inconvenience but the installers got broken

we are currently working on Debian 12 packages

you could find them here

https://repository.qtpyvcp.com/repo/qtpyvcp-dev/
https://repository.qtpyvcp.com/repo/probe-basic-dev/

hasechris, 2023-04-02:
Hi @TurBoss :)

firstly thank you so much for this work here. Probe basic is for me the only solution to a usable linuxcnc installation. So dont feel to bad, i was also a bit stupid to update my installation before wanting to mill something :D.

> we are currently working on Debian 12 packages
Hm now im puzzled. Currently i cant find a linuxcnc installation iso with PREEMPT-rt kernel which is based on Debian 11, let alone Debian 12. Why are you working on Debian 12 packages? Can you give me some hints? Im currently on LinuxCNC 2.8 (http://linuxcnc.org/downloads/) which is based upon Debian 10.

Best regards and greetings from Germany
hasechris

flowers84, 2023-04-12:
Hi. '
that is interesting ! Do you use Master branch (development - 2.9) in Debian 12?

All the best
Stefan

TurBoss, 2023-04-12:
yes, I have a debian 12 virtual machine to test all the features and just installed linuxcnc from apt 

I had to install linuxcnc-uspace + linuxcnc-uspace-dev to get all the requirements 

Saludos! from spain

flowers84, 2023-04-12:
Since debian 12 is being worked on, does the whole thing also work on debian 11?

Thanks

TurBoss, 2023-04-12:
yes but only if you install it from repositories and do a `pip install -e .` + `qcompile .`

Lcvette, 2023-04-23:
> Since debian 12 is being worked on, does the whole thing also work on debian 11?
> 
> Thanks

I would not bother with debian 11 it is a stepchild of sorts.  easiest to use debian 12 and install linuxcnc from apt and probe basic/qtpyvcp from .deb file for user use.  and for future updates, the quickstart page will have updated deb file which are installed over the existing very easy one line update and easy to see the version number currently listed in the installation docs.

https://kcjengr.github.io/probe_basic/quick_start.html

hasechris, 2023-04-23:
@Lcvette 
@TurBoss

wait WHAT? 
Why do you suddenly drop support for the "official" LinuxCNC ISO? The ISO uses Debian 10 with LinuxCNC 2.8.4.

Could we please get back to support the LinuxCNC ISO? I imagine most normal users dont bother with a manual Debian configuration and setup and "just want to mill" something. So they use the linuxcnc iso with the assumption "this is the stable build and should work in the normal cnc work condition".

Until now this was just a nuance that you broke the installer for debian 10. Now you suddenly inform us that this is a breaking change and we have to reinstall all of our existing CNC Computers.

Best regards
hasechris

TurBoss, 2023-04-29:
hello, sorry for the late reply.

i found this binaries

https://repository.qtpyvcp.com/repo/dev/bin/
https://repository.qtpyvcp.com/repo/pb-py2/bin/

hope they work

hasechris, 2023-05-20:
Hi @TurBoss,

sadly i could not test your links until now. Sorry that it took me this long.

First thank you for providing the (in your mind) old run files. That enabled me to run the normal probe_basic installer on my LinuxCNC ISO Install version 2.8.4 (which is still up to date as of 2023-05-20). 

Sadly an error came up in the middle of the installation. Please see the new mentioned issue (https://github.com/kcjengr/qtpyvcp_conversational_gcode/issues/7) in the conversational repo. This repo want to be installed by the run file.

Workaround for now is to do
```
pip install git+https://github.com/kcjengr/qtpyvcp_conversational_gcode.git@1b7273b5d227867b76e9cd2d658a414f2935a12a
./ProbeBasic-Installer-py2+3.g231c7ff.run
```
** AND press the ignore Button when the error regarding the conversational repo comes up in the run-file installer**

Best regards
hasechris