kcjengr/probe_basic #77: Expand wheel widget to 24 or 36 tools

Stephanisk, 2023-07-31:
Would it be possible to extend the wheel widget to hold more than 12 tools. It would be great if you could set yourself the number of toolholders in your ATC. For example between 6 and 36.

dpslwk, 2023-08-01:
24 pockets are already supported along with a number of other sizes, 

https://github.com/kcjengr/probe_basic/blob/b9bc979adbbde28a1464816a547554d91fdd0cd1/configs/probe_basic/probe_basic.ini#L50-L53

adding others would at a minimum need more images making and I guess some code tweaks
https://github.com/kcjengr/probe_basic/tree/main/src/widgets/atc_widget/images

Stephanisk, 2023-08-27:
I am close to provide the linuxcnc forum with a fully parametrical and adaptive design for a professional toolchanger. The number of tools is basically unlimited. Though practical I believe 36 seems about the maximum. It would be really great if this could be implemented in probe basic. ( 26-28-30-32-34-36). This way this design would be highly functional for all our DiY guys out there together with probe basic