kcjengr/probe_basic #99: "Spindle speed display rpm source" setting not remembered

dpslwk, 2023-11-26:
On the setting page we have "Spindle speed display rpm source" section with two buttons in the `spindlerpmsourcebtnGroup` group

* "CALCULATED SPINDLE RPM" `spindle_calculated_rpm`
* "ENCODER SPINDLE RPM" `spindle_encodr_rpm`

these change the page of `spindle_rpm_source_widget` via some python (left spindle rpm box)

but although these are `VCPSettingsPushButton`'s they have no `settingName` set and we are not remembering this state in anyway, so at launch PB always goes back to encoder rpm and not everyone has an encore or it  hooked up in hal

Lcvette, 2023-12-15:
spindle speed is now fixed, link angular jog is not yet completed, need to look into that a bit more to see how the best way to resolve the connection would be.

Lcvette, 2023-12-15:
I am going to close this issue but let open one just for the link angular jog issue to tidy up the fix list.

closed 2023-12-15 by Lcvette