kcjengr/probe_basic #118: Menu option to override limits is not working

Unlogic-dot-se, 2024-03-29:
I'm unable to get the menu option for overriding limits to work.

It's visible in the top menu but nothing happens when I click on it.

I found a post from 2021 on the LinuxCNC forum describing the same issue:

https://forum.linuxcnc.org/qtpyvcp/42675-probe-basic-using-override-limits#210341

Unlogic-dot-se, 2024-04-05:
I did some more digging into this issue and discovered there is actually two ways to trigger the "Limits override" function in Probe Basic.

One way is using the window menu (which doesn't seem to work), the other way is to toggle the button under the WCS tab on the right side and that way it seems to work fine.

[image]

TurBoss, 2024-04-05:
Hello,

thanks for the report! I'm taking a look