kcjengr/probe_basic #124: Update Probe/Toolsetter Parameters not working

NoEngineerHere, 2025-06-06:
Pressing the update parameters button has no effect, both in previously working configuration and in included probe basic sim config (with fresh linuxcnc install). 

Also unable to type values into the traverse feedrate, fast feedrate, and X/y clearance fields - they revert to 0 when clicking out or pressing enter.

Lcvette, 2025-06-06:
confirmed, this is a PITA issue with QTDesigner, its tied to the line edit text format and for some reason loses its setting when pushing updates sometimes but not others, I will fix now, should have an update in 15 minutes or so pushed to resolve. I will reply here when its up and ready to run apt update!

NoEngineerHere, 2025-06-06:
Legend, sounds like I'll be making chips in no time! 

Thanks for your diligence.

closed 2025-06-06 by Lcvette

NoEngineerHere, 2025-06-06:
Just gave it a go - text fields and tool setter parameter update button work fine - but update Probe parameters still not functioning.

Lcvette, 2025-06-06:
> Just gave it a go - text fields and tool setter parameter update button work fine - but update Probe parameters still not functioning.

i found.. missed one of the param entry boxes that soured, the hints.. forgot to test those fixinf now will push in a moment

Lcvette, 2025-06-06:
fixed now ! sorry about that

NoEngineerHere, 2025-06-06:
No need to apologise - you're killing it. Tested and working on my end!