kcjengr/probe_basic #125: Identifying where the configuration is going wrong?

JonHodgson, 2025-08-01:
After updating linuxcnc and probe_basic I am getting an "Important: Please Read, Action Required!" message telling me the config needs modifying.
I've tried updating according to the instructions, which are rather sparse, and so far it is not helping. But the bigger issue is that there seem to be absolutely no clues as to what specific things are failing, so I have absolutely no clues to help me to troubleshoot the issues.

Is there some useful log output somewhere?

Lcvette, 2025-08-03:
the docs go over absolutely everything you need to know about configuration requirements for probe basic. if the sim works opens and doesn't complain the installation is correct, if the sim has issues then there is an issue with your installation. here are the docs:

https://kcjengr.github.io/probe_basic/

JonHodgson, 2025-08-03:
It's all very well saying the docs cover everything, but people make mistakes, it may be a typo, it may be a misunderstanding, or they might have missed something (especially when inheriting a configuration from someone else, as in my case).
And when that happens, a "there is an error somewhere in your configuration but we're not even going to give you a clue as to what it is or where it is" message isn't really very helpful is it?
Which is why I asked if there was a log.

Lcvette, 2025-08-03:
yes there is a log, it should be created and updated in your config folder every time you start linuxcnc.  it will have the error messages or failure points of the last instance you attempted to run.  it will be:

your_name_you_used.log

closed 2026-02-12 by Lcvette